samitbasu/rust-hdl #2: Inline signed conversions don't compile.

john-terrell, 2022-12-25:
I'm attempting to perform a signed comparison between two Signals containing Bits types.   The following doesn't compile in the latest build:

if self.a.val().to_signed_bits() < self.b.val().to_signed_bits() {
    ...
}

The compiler complains: 

#[cfg(not(any(feature = "foo", feature = "bar")))]
compile_error!("Either feature \"foo\" or \"bar\" must be enabled for this crate.");
Unsupported method call for hardware conversion

samitbasu, 2023-01-01:
Ok - the problem is that there is support for a `Signed` data type in the HDL generation, but not for signed casts.  Adding support for those now.

samitbasu, 2023-01-01:
Ah - ok.  So the first issue is that for conversion from `Bits -> Signed`, you want the functional form `signed_cast`.  So you should have:

```rust
if signed_cast(self.a.val()) < signed_cast(self.b.val()) {
...
}
```

However, that does not work either (the comparison ops are missing).  So I need to fix that as well.

samitbasu, 2023-01-02:
Hi @john-terrell  - I put (what I believe are the needed fixes) on a branch.  Signed arithmetic still needs some TLC.  But if I can get the test suite to pass, I'll merge it to master and publish a new version.

closed 2023-01-02 by samitbasu