samitbasu/rust-hdl #5: Question about the generated verilog file

AdiwenaPutra, 2023-01-02:
Hello, I have been checking your code from the documentation and I got until this point:
[image]

Running that code will give you following verilog results:
[image]

My concern is, if you go to top module file and see the `// Update code` part, you will se `my_reg$d` register has been drived by two wires. Isn't this cause conflict? Thank you

samitbasu, 2023-01-02:
Hello @AdiwenaPutra ,

It's OK to have to assigns to the same register as Verilog will take the last one as the highest "priority".  The first one is basically there for latch prevention.  

Indeed, the way RustHDL generates code is to make it easier to reason about the assignment priority.  You can make as many assignments to a `.next` endpoint as you want. Only the last one that is executed really matters.

Samit

AdiwenaPutra, 2023-01-03:
alright, thank you for the explanation

closed 2023-01-03 by AdiwenaPutra