samitbasu/rust-hdl #7: Logical shift left using Bits<> of different widths doesn't compile

john-terrell, 2023-01-03:
E.g.  This fails to compile:

self.result.next = self.a.val() << self.b.val().get_bits::<5>(4);

samitbasu, 2023-01-05:
Same as the previous issue.  You will need the two arguments to `<<` to have the same bit width.  So you can do:

```
self.result.next = self.a.val() << bit_cast::<M,N>(self.b.val().get_bits::<5>(4);
```

I think you can also elide the sizes for `bit_cast`, so that it's

```
self.result.next = self.a.val() << bit_cast::<_,_>(...)
```

In retrospect, I should probably add implementations of the left and right shift for arbitrary bit-width pairs, but in general, the number of bits in the shift are probably less than `log2(M)`, where M is the number of bits in the value being shifted.  So I just upper bounded `log2(M) < M`.  If you feel this should be improved (or better documented), I can add it.

Samit

samitbasu, 2023-01-08:
I added this capability to `0.43.0`.  Please update and this should be fixed.

closed 2023-01-08 by samitbasu