samitbasu/rust-hdl #8: Arithmetic shift right not compiling

john-terrell, 2023-01-03:
Attempting to perform an arithmetic shift right fails to compile.  Is there a better way to handle this?

                self.result.next = signed_cast(self.a.val()) >> self.b.val();

samitbasu, 2023-01-05:
I currently only support right and left shift if the two arguments are of the same bit width.  So you should be able to do something like:

```
self.result.next = a >> bit_cast::<M,N>(b);
```

This will satisfy Rust.  Can you try that?

samitbasu, 2023-01-08:
Ok - I added support for `A << B` where `A` and `B` are different bit widths.  No support for signed ops in this branch, but unsigned ops should work.  The performance won't be great, but the flexibility is probably more important.

closed 2023-01-08 by samitbasu