samitbasu/rust-hdl #12: Support for multidimensional packed bit array

BinhaoQin, 2023-03-15:
In system verilog, I usually write things like `logic [2:0] [1:0]`, and I'm wondering is that possible for RustHDL.

samitbasu, 2023-03-15:
Hi - not yet (or at least I haven't tried it).  I think we need a new class for that like `BitArray2D` or equivalent.  It's a good idea though.  In Rust, this would probably generic over both the number of rows and columns.  Something like this:

```rust
pub struct BitArray2D<const N: usize, const M: usize> {
    value: [Bits<M>;N]
}
```

You might be able to just declare a signal of type `[Bits<M>; N]` and it may just work.  Also, you should be able to do something like:

```rust
   my_signal: Signal<[Bits<M>;N], In>
```
In which case, you should be able to do something like

```rust
my_signal.val()[n].get_bits(...)
```
But I don't know if it will be processed correctly. 

Also, you may be looking for more complex operations, like extracting a 2D subset of bits out of larger bit array.  That would require some kind of generator support in the Verilog backend, and that definitely doesn't exist.

Let me know your thoughts.  It's definitely a good feature to think about adding.

samitbasu, 2023-03-17:
We certainly can. I think the only difficulty is supporting the various indexing and assignment syntaxes in Verilog.

BinhaoQin, 2023-03-18:
In fact, multidimensional packed array is only supported in SystemVerilog. Thus, instead of generating to Verilog, I'm more interested in developing a backend for SystemVerilog. I'm wondering if you are also interested.

samitbasu, 2023-03-20:
Yeah, @BinhaoQin - I am definitely interested.  My experience with SV was that toolchain support was pretty uneven.  Some of the parts I worked with are old (Spartan6) and the toolchains had pretty poor SV support.  But I don't object to it as a synthesis target.

BinhaoQin, 2023-03-23:
Thanks, @samitbasu. I'm interested in this. Maybe we can discuss how this works out.

samitbasu, 2023-03-24:
Yeah, totally!  I think we have a [discord](https://discord.gg/qcK3dkCG) set up if you want to try it out...