samitbasu/rust-hdl #22: Incorrect Verilog Output

mwbryant, 2023-04-25:
I'm probably not doing things in the most idiomatic way yet but I'm reading from some rams based on the x and y location on the screen for a vga text display.  

I'm doing some bit magic to get the addresses and the behavior worked in the build in simulation but upon generating and flashing the verilog file the results were incorrect and different from the sim.

 ```rust 
self.character_ram.read_address.next = bit_cast(
            bit_cast(self.screen_x.val() >> 4) + bit_cast(self.screen_y.val() >> 4) * 40.into(),
        );

        self.font_ram.read_address.next = bit_cast(
            (bit_cast::<17, 8>(self.character_ram.read_data.val()) << 8)
                | bit_cast(self.screen_y.val().get_bits::<4>(0)) << 4
                | bit_cast(self.screen_x.val().get_bits::<4>(0)),
        );
```

```verilog
// Generated 
character_ram$read_address = screen_x >> 32'h4 + (screen_y >> 32'h4 * 32'h28);
font_ram$read_address = (character_ram$read_data << 32'h8) | screen_y[(32'h0)+:(4)] << 32'h4 | screen_x[(32'h0)+:(4)];
// Corrected by hand
character_ram$read_address = screen_x[9:4] + (screen_y[8:4] * 32'h28);
font_ram$read_address = {character_ram$read_data , screen_y[3:0], screen_x[(32'h0)+:(4)]};
```

samitbasu, 2023-04-25:
It's definitely a bug. I think the multiplication was dropped and I'm not sure why.

samitbasu, 2023-04-26:
Ah!  I think it's an order of operations issue. The multiply and the right shift have inverted priorities.

samitbasu, 2023-04-30:
Since I am swamped at work at the moment can you try putting some parentheses into the expression to test my hypothesis? @mwbryant

samitbasu, 2023-07-01:
Ok - let's see what's going on here.  Here is the first line:

```
self.character_ram.read_address.next = bit_cast(
 bit_cast(self.screen_x.val() >> 4) + bit_cast(self.screen_y.val() >> 4) * 40.into(),
);
```

The first one is basically `addr = x / 16 + y/16 * 40`.  The generated code is

```
character_ram$read_address = screen_x >> 32'h4 + (screen_y >> 32'h4 * 32'h28);
```

which looks similar... However, there is a difference between the Verilog and the Rust context - the `bitcast` operator will mask off higher order bits.  And Verilog will not.  So my suspicion is that the difference comes down to the upper bits of `screen_y` and `screen_x`.  In particular, let's consider a more basic case where we have a constant like:

```
    let x: Bits<8> = 0xFF.into();
    let y: Bits<6> = bit_cast(x);
```
In RustHDL, this generates a value of `0x3F`.  But in verilog, it does not.  That is because in Verilog, the bit_cast operation is a no-op.  Yuck.  

 I will fix this.

samitbasu, 2023-07-02:
It turned out to relate to the specifics of the expression above and the use of nested `bit_cast` without specifying the destination type, along with a missing mask in the case of bit casting.    It's fixed in 0.46.0

closed 2023-07-02 by samitbasu