samitbasu/rust-hdl #23: Support Sum Types

ThePerfectComputer, 2023-05-11:
Both Clash HDL and Bluespec have support for sum types. If rust-hdl could have sum types, where a register could simply be sized to hold the largest variant of the sum type - that could be a huge productivity boon in terms of designing things like decoders.

I do see [this](https://github.com/samitbasu/rust-hdl/blob/4672ca271dac3aee771eed310d198f4f70c55631/rust-hdl-core/src/module_defines.rs#L29), perhaps that's where one could begin to work towards implementing sum types?

ThePerfectComputer, 2023-05-11:
They're a particular team at a well known company that may be able to use RustHDL in production. I do think however that Sum Types could really help.

samitbasu, 2023-05-11:
That is a good point. I never thought of them for hardware designs. But I don't see a priori why you could not add them. There is already the notion of both enums and structs. I will work on some concepts. Thanks for the suggestion!