samitbasu/rust-hdl #25: rust-hdl and circt

jcdutton, 2023-06-25:
Would it be possible to have rust-hdl output to circt.  https://circt.llvm.org/
One could then use the LLVM/MLIR to optimize the output before outputting System Verilog or VHDL.
For example rust is already compiled via LLVM, why not compile rust-hdl via CIRCT ?
If one tries to create Verilog from rust-hdl directly, one kind of skips the optimization opportunities.

samitbasu, 2023-07-01:
@jcdutton - it looks interesting, but the documentation is a bit hard to follow.  Do you have some pointers to how to actually use `circt` to build something?

jcdutton, 2023-07-02:
As you can see from the diagram on the  https://circt.llvm.org/.  There are several routes to get to FPGA through circt.
I would suggest looking at "Chisel" and "polygeist" as starting points.
Chisel lowers to firrtl, so maybe lowering rush-hdl to firrt would suit.
polygeist lowers to MLIR SCF, AFFINE, ARITH, CF.  So if rush-hdl is more like polygeist, lowering the same way polygeist does might be best.
The Polygeist path permits many more higher level optimizations than the firrtl path.
If you try an Chisel example, and then a polygeist example and see what you find.