samitbasu/rust-hdl #30: Calling `Simulation::add_testbench` makes the simulations interfere with each other

PoignardAzur, 2023-07-23:
Minimal example:

```rust
#[test]
fn test() {
    let mut sim = Simulation::new();
    // TEST_A
    sim.add_testbench(move |mut endpoint: Sim<NandGate>| {
        let mut x = endpoint.init()?;
        x.input_1.next = true.into();
        x.input_2.next = true.into();
        let mut x = endpoint.wait(10 * sim_time::ONE_MICROSECOND, x).unwrap();
        sim_assert_eq!(endpoint, x.output_signal.val(), false, x);
        endpoint.done(x)
    });
    // TEST_B
    sim.add_testbench(move |mut endpoint: Sim<NandGate>| {
        let mut x = endpoint.init()?;
        x.input_1.next = false.into();
        x.input_2.next = true.into();
        let mut x = endpoint.wait(10 * sim_time::ONE_MICROSECOND, x).unwrap();
        endpoint.done(x)
    });
    sim.run_to_file(
        Box::new(NandGate::default()),
        5 * sim_time::ONE_SEC,
        "nandgate.vcd",
    )
    .unwrap();
}
```

(note the absence of clocks; I don't know if it's a factor)

When the tests are added in the order above, the simulation hangs instead of succeeding. If they're added in the opposite order, (TEST_B then TEST_A), the simulation succeeds.

If TEST_A is changed to fail, there are two scenarios:
- If TEST_A is added before TEST_B, the simulation succeeds.
- If TEST_B is added before TEST_A, the simulation fails.

Basically the simulation only "remembers" the last test.

samitbasu, 2023-07-23:
I don't think this is a problem of multiple test-benches, I think the problem is that the two testbenches are "fighting" over the inputs to your circuit.  From what I can see, during the first 10 microseconds, both testbenches are driving the same inputs of the circuit to _different_ values.  In a physical circuit this would cause bad things to happen.  I don't think I can detect this condition, since it is hard to enforce the single-writer principle in the testbenches.  

Does that make sense?  In RustHDL, test-benches execute on the same circuit simultaneously.  If you want to have 2 separate test cases, you need to run 2 separate simulations.

samitbasu, 2023-07-23:
Alternately, if you want to run multiple test benches _sequentially_ in the same simulation, you need to make the second one wait until the first one is done (by adding a 10 microsecond wait at the beginning, for example), or just putting both test cases into a single testbench and executing them sequentially.