samitbasu/rust-hdl #31: Dumping BRAM Contents in Simulation

ThePerfectComputer, 2023-10-02:
Is there a way to dump BRAM contents during simulation? For example, I may want to step simulation one cycle, and then print BRAM contents - and repeat this process.

If this is possible, how might I go about it?

samitbasu, 2023-10-02:
I can think of two ways to do this.

1.  You can trace your simulation.  The resulting `.vcd` file will include the BRAM contents, which means you can see it in something like `gtkwave`.
2. You can add debug statements to your simulation and just print out the BRAM while it is running.  This tends to be a little hard to deal with because each logic routine gets called multiple times in a clock cycle as the logic changes propagate through the network.

LMK if that helps.
Samit

ThePerfectComputer, 2023-10-02:
Sounds useful. I guess I'm also wondering how to instantiate a BRAM - couldn't find any example for this.

ThePerfectComputer, 2023-10-02:
Also - before switching to RustHDL, I'm wondering if I can probe signals nested in a module hierarchy. VHDL and Verilog support for this is pretty cumbersome requiring "bubbling up" of probes.

samitbasu, 2023-10-03:
For the first one, I would suggest using the BRAM widget, as defined [here](https://github.com/samitbasu/rust-hdl/blob/main/rust-hdl-widgets/src/ramrom/ram.rs).  An example of logic using the BRAM is [here](https://github.com/samitbasu/rust-hdl/blob/main/rust-hdl/tests/core_ram.rs).

For the second - yes!  RustHDL simulations are hierarchical, and support automatic inclusion of the entire hierarchy with no additional work on your part.  I think if you run the integration tests, you get several examples of deeply nested VCDs you can examine.

Note that this is slightly less true in `rhdl`.  There, you need to at least flag values that you want to include in the traces.   But once flagged, they are composed into hierarchical logs as you would expect.  I chose this path since in many cases, a lot of the logging is not only superfluous, but can be so overwhelming that it's hard to find the signals that you actually want.  And a multi-GB vcd file can be a pain.