samitbasu/rust-hdl #41: Support for Sipeed Tang Nano 9K FPGA Development Board Gowin GW1NR-9 RISC-V HDMI (Tang Nano 9k)

Toni-Graphics, 2024-04-04:
Does your cool libary support the [Sipeed Tang Nano 9K FPGA Development Board Gowin GW1NR-9 RISC-V HDMI (Tang Nano 9k)](https://www.amazon.com/-/de/dp/B0BCXYWV3T/ref=pd_vtp_h_pd_vtp_h_d_sccl_1/131-0425290-8694127?pd_rd_w=ecLXB&content-id=amzn1.sym.e16c7d1a-0497-4008-b7be-636e59b1dfaf&pf_rd_p=e16c7d1a-0497-4008-b7be-636e59b1dfaf&pf_rd_r=0XWEP4GW9E436RWQRV6T&pd_rd_wg=ktIMl&pd_rd_r=6a64d982-a17a-4bba-b885-c3a4f1cc1339&pd_rd_i=B0BCXYWV3T&th=1)? <- In the name is a link to the board

Bye

Toni-Graphics, 2024-04-07:
@samitbasu

samitbasu, 2024-04-08:
@Toni-Graphics - there is no reason _not_ to support it.  Someone would need to craft a board support package, like this [one](https://crates.io/crates/rust-hdl-bsp-step-mxo2-lpc).   I am unfamiliar with the toolchain for this FPGA.  But there are BSPs for several other toolchain/FPGA combos, so I don't see any fundamental issue.

samitbasu, 2024-04-08:
The documentation I could find is in Chinese - Google translates helps - a little.  But I'm definitely out of my depth on this one.  It looks like a great FPGA, though.

Toni-Graphics, 2024-04-09:
Cool

parker-research, 2024-05-10:
Would be awesome if this got implemented! The Gowin boards are very, very affordable, and seem to be growing pretty rapidly in popularity.