samitbasu/rust-hdl #46: State machine that doesn't work

OscarLopezCentenera, 2025-04-16:
Hi, I'm writing because I'm trying to implement a pattern recognizer in Rust that generates code in Verilog and at the same time I can run the test with the command "cargo test -- --nocapture". But the machine isn't updating the state and I've been struggling with this for easily 20 hours and I can't get it to work. I'm attaching the main.rs code and what the command returns to see if you can help me. I think it's not updating the state correctly and most likely I'm creating the state machine incorrectly but it could also be that it doesn't support the creation of it. I'm waiting for a response.

----------------------Code Rust --------------------------------
use rust_hdl::prelude::*;
use std::fs;
use std::fs::File;
use std::io::Write;
use anyhow::anyhow;

#[derive(Copy, Clone, PartialEq, Debug, LogicState)]
enum State {
    S0, // Estado inicial
    S1, // Detecta '1'
    S2, // Detecta '10'
    S3, // Detecta '101'
    S4, // Detecta '1011' (estado de éxito)
}

#[derive(LogicBlock, Clone)]
struct ReconocedorPatrones {
    pub clk: Signal<In, Clock>,
    pub rst: Signal<In, Bit>,
    pub entrada: Signal<In, Bit>,
    pub salida: Signal<Out, Bit>,
    state: DFF<State>,
}

impl Default for ReconocedorPatrones {
    fn default() -> Self {
        let mut instance = Self {
            clk: Default::default(),
            rst: Default::default(),
            entrada: Default::default(),
            salida: Default::default(),
            state: DFF::default(),
        };
        instance.state.d.next = State::S0;
        instance
    }
}

impl Logic for ReconocedorPatrones {
    #[hdl_gen]
    fn update(&mut self) {
        dff_setup!(self, clk, state); //Conecta el DFF al reloj

        if self.rst.val() {
            //Si activas el reset volemos al estado inicial
            self.state.d.next = State::S0;
            self.salida.next = false;
        } else {

            println!(
                "Estado actual: {:?}, Entrada: {}, Salida: {}",
                self.state.q.val(), self.entrada.val(), self.salida.val()
            );

            match self.state.q.val() {
                State::S0 => { // Estado S0
                    println!("En S0, Entrada: {}", self.entrada.val());
                    if self.entrada.val() {
                        println!("S0 dentro");
                        self.state.d.next = State::S1;
                    }
                    else {
                        self.state.d.next = State::S0;
                    }
                    self.salida.next = false;
                }
                State::S1 => { // Estado S1
[... 270 lines cut ...]
En S1, Entrada: true
Estado actual: S0, Entrada: false, Salida: false
En S0, Entrada: false
Estado actual: S0, Entrada: true, Salida: false
En S0, Entrada: true
S0 dentro
Estado actual: S0, Entrada: true, Salida: false
En S0, Entrada: true
S0 dentro
Entrada: 1, Iteración: 4, Salida esperada: true, Salida actual: false
HALT x.salida.val() != expected_output,  false != true
Estado actual: S0, Entrada: true, Salida: false
En S0, Entrada: true
S0 dentro
Error: SimHalted
FAILED

failures:

failures:
    test_reconocedor_patrones

test result: FAILED. 0 passed; 1 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.02s

error: test failed, to rerun pass `--bin reconocedorPatrones`

samitbasu, 2025-04-25:
Hi @OscarLopezCentenera  - do you have the code checked in somewhere?  I can't really follow from the pieces posted above.  I could try and clone your repo and see if I can understand what is happening.

OscarLopezCentenera, 2025-04-25:
No problem, I've uploaded my codes to Drive and I'm attaching the link. I've been using the "cargo run" and "cargo test --nocapture" commands in case you want to replicate the outputs.
https://drive.google.com/drive/folders/1gnTaJIIlN-YGZId5HlkXbbe4yR0BxnYsxTCRU59KeuBPoDnA9W8-YEF1Cc7OFK-e7rjX5GzZ?usp=sharing
If you need anything because you see that you can't access the drive, write to me at lopezcenteneraoscar@gmail.com