samitbasu/rust-hdl #48: Compilation error for `rust-hdl-ok-frontpanel-sys`

loriiiiiiiii, 2025-05-15:
Environment:
- Windows 11 24H2
- rust-hdl-ok-frontpanel-sys v0.46
- FrontPanel SDK v5.3.6

Error msg:
```
error[E0308]: mismatched types
      --> C:\Users\LoriLi\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\rust-hdl-ok-frontpanel-sys-0.46.0\src\lib.rs:129:62
       |
129    |         unsafe { okFrontPanel_SetWireInValue(self.hnd, addr, val as u64, 0xFFFF) };
       |                  ---------------------------                 ^^^^^^^^^^ expected `u32`, found `u64`
       |                  |
       |                  arguments to this function are incorrect
       |
note: function defined here
      --> C:\Users\LoriLi\Documents\GitHub\hdl_test\target\debug\build\rust-hdl-ok-frontpanel-sys-4c239c464127879b\out/bindings.rs:177673:12
       |
177673 |     pub fn okFrontPanel_SetWireInValue(
       |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
...
177676 |         val: ::std::os::raw::c_ulong,
       |         ---
help: you can convert a `u64` to a `u32` and panic if the converted value doesn't fit
       |
129    |         unsafe { okFrontPanel_SetWireInValue(self.hnd, addr, (val as u64).try_into().unwrap(), 0xFFFF) };
       |                                                              +          +++++++++++++++++++++

For more information about this error, try `rustc --explain E0308`.
error: could not compile `rust-hdl-ok-frontpanel-sys` (lib) due to 1 previous error
```

Related file:
okFrontPanel.h
```
okFrontPanel_SetWireInValue(okFrontPanel_HANDLE hnd, int ep, unsigned long val, unsigned long mask);
```

loriiiiiiiii, 2025-05-15:
When I changed the source code `u64` into `u32`, the program compiles without problem. The frontpanel SDK might have changed its interface slightly without telling anyone...

loriiiiiiiii, 2025-05-15:
The sdk also has a different folder structure, making it incompatible with the current `build.rs`. Currently, the build system assume that the .h file and the .lib file are both under the same `API\` folder. However, the actual folder structure looks like this: 
\path\to\Opal Kelly\FrontPanelUSB\API\
├── include\
│   └── okFrontPanel.h
├── lib\
│   └── x64\
│    |   └── okFrontPanel.lib
This mismatch cause the linker to produce the following error message:
```
error: linking with `link.exe` failed: exit code: 1181
  |
  = note: "C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.43.34808\\bin\\HostX64\\x64\\link.exe" "/NOLOGO" "C:\\Users\\LoriLi\\AppData\\Local\\Temp\\rustcF7YUbw\\symbols.o" "<8 object files omitted>" "C:\\Users\\LoriLi\\Documents\\GitHub\\hdl_test\\target\\debug\\deps/{librust_hdl_ok_frontpanel_sys-444eeae01ccd15dd.rlib}.rlib" "<sysroot>\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib/{libstd-*,libpanic_unwind-*,libwindows_targets-*,librustc_demangle-*,libstd_detect-*,libhashbrown-*,librustc_std_workspace_alloc-*,libunwind-*,libcfg_if-*,liballoc-*,librustc_std_workspace_core-*,libcore-*,libcompiler_builtins-*}.rlib" "okFrontPanel.lib" "kernel32.lib" "kernel32.lib" "advapi32.lib" "ntdll.lib" "userenv.lib" "ws2_32.lib" "dbghelp.lib" "/defaultlib:msvcrt" "/NXCOMPAT" "/LIBPATH:C:\\Program Files\\Opal Kelly\\FrontPanelUSB\\API\\lib" "/OUT:C:\\Users\\LoriLi\\Documents\\GitHub\\hdl_test\\target\\debug\\deps\\hdl_test.exe" "/OPT:REF,NOICF" "/DEBUG" "/PDBALTPATH:%_PDB%" "/NATVIS:<sysroot>\\lib\\rustlib\\etc\\intrinsic.natvis" "/NATVIS:<sysroot>\\lib\\rustlib\\etc\\liballoc.natvis" "/NATVIS:<sysroot>\\lib\\rustlib\\etc\\libcore.natvis" "/NATVIS:<sysroot>\\lib\\rustlib\\etc\\libstd.natvis"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: LINK : fatal error LNK1181: cannot open input file 'okFrontPanel.lib'␍
```

samitbasu, 2025-06-03:
Is it worth trying to make the fix backward compatible to older versions of Front panel?